uf-mil/mil2 #15: Make C++ `electrical_protocol`

cbrxyz, 2025-01-17:
We should have a C++ version of the `electrical_protocol` package so that we're not restricted to only using Python for our electrical drivers.

I found this cool cpp library that parallels `struct` in Python: https://github.com/karkason/cppystruct. It has `constexpr` support, too!

zrzz-hq, 2025-02-10:
This week, I finalized the software design plan for electrical_protocol c++ version, completed the SerialDevice class, and am currently writing the test code. Commits are on the electrical_protocol branch.

zrzz-hq, 2025-02-11:
The development of ringBuffer as part of the CTest framework for testing the SerialDevice class has been completed.

zrzz-hq, 2025-02-17:
The C++ version of the electrical protocol has passed all CTests and is now ready for testing on real devices.

zrzz-hq, 2025-02-18:
Which node will use the electrical protocol? I'm going to rewrite that node to test the SerialDevice class.