uf-mil/mil2 #25: Gazebo: Add the thrusters plugin

cbrxyz, 2025-01-30:
Several simluators have plugins that we could use!

lucasperez-techdev, 2025-02-10:
This week I worked on the Gazebo branch to implement the thruster plugin functionality. Most of my time was researching how xacro  files worked/how gazebo plugins work, how to integrate the gz-sim thruster plugin into our simulation and looking through other projects like Dave and UUV_Sim to gain some knowledge on how other people implemented similar plugins. I created the initial template for the plugin but was not able to get it fully functional however Kieth and Carter were able to fix it up I believe.

CarterKreis, 2025-02-10:
This week Lucas, Keith, and I worked on the thrusters plugin for Gazebo. We looked through Gazebo documentation and online examples of the Gazebo thruster plugin such as Dave and Gazebo's provided examples. We ran into on properly formatting the xacro file but after looking at provided examples from gz-sim and a few typos later Keith and I finished up on Lucas's work for the plugin.

CarterKreis, 2025-02-14:
Confirmed Thruster orientations by simulating in Gazebo and testing each one individually

closed 2025-02-14 by CarterKreis