uf-mil/mil2 #26: Gazebo: Add depth sensor/pressure sensor plugin

cbrxyz, 2025-01-30:
Several simluators have plugins that we could use!

Carlosdc25, 2025-02-10:
This week, I started working on adding a depth sensor plugin and mostly just started learning how to make gazebo plugins. Additionally, I've made a template for the plugin using the previous depth sensor plugin used in the old mil repo. On a different note, this week, I added a wifi driver so I can run wifi on my machine(finally!). Lastly, I have been keeping up with the ROS2 tutorials.

DaniParr, 2025-02-10:
- [ ] Screenshot of the plugin printing to the screen/publishing to a topic

Carlosdc25, 2025-02-17:
This week I continued to work on the depth sensor plugin for subjugator. I was able to successfully build the depthSensor.cc file and then I created an xacro file for testing.  The placement of the sensor isn't well thought out because currently im just testing to see if it works. Currently, I'm not able to get the sensor to print to the console, so I am working on getting that working.

Carlosdc25, 2025-02-24:
This week with Keith's help I was able to publish sub9's depth to the depth topic. Spent some time learning about and sensor and model plugins in Gazebo. The time that was being published to the depth topic was in nanoseconds so it was a really large number and not very helpful. I made that time into seconds and nanoseconds. Finally, I added Gaussian noise to the depth measurement to try to mimic unstable conditions in the real world. In the first photo, i am publishing depths of the sub floating at the surface and the noise has a standard distribution of 0.1. The second photo has a std of 0 under the same conditions.

[image]
[image]