uf-mil/mil2 #27: Gazebo: Add active sonar plugin

cbrxyz, 2025-01-30:
Several simluators have plugins that we could use!

alexoj46, 2025-02-04:
Project DAVE might have something useful: https://github.com/Field-Robotics-Lab/dave/blob/master/Changelog.md

We are interested in simulating something close to the Ping360 single-beam sonar.

ElijaRock, 2025-02-14:
Keith assigned Oickle and I to take this issue for the following week (next week) of creating the active sonar plugin.

ElijaRock, 2025-02-18:
Started the Active Beam Sonar Gazebo plugin. We are going to create a beam in Gazebo and then spin it around to simulate an active sonar beam.

Data from real device (Ping 360): https://github.com/CentraleNantesRobotics/ping360_sonar/blob/ros2/ping360_sonar_msgs/msg/SonarEcho.msg

If gazebo cannot provide collision information from beams as well as angles with distance, we can also try calculating the distance to each object from a single point and then "pretend" that we have a sonar (it will perform the same function less efficiently).

ElijaRock, 2025-02-21:
We need to figure out two things. There is a collision for the sonar, however there is no rotation on it now or data to collect from it. In terms of moving it in real time, there are other objects that can be referenced that do move like springs for example (http://sdformat.org/tutorials?tut=sdformat_urdf_extensions&cat=specification&#gazebo-elements-for-robot). There are also other tags that can be tested in geometry, etc, if they can be found online.

In terms of retrieving information from the collision, we are not sure if that is possible at the moment which means we may have to find a workaround for that.

oixel, 2025-02-21:
Additionally, SetForce or GetJoint might be helpful for possibly grabbing the joint the collision is attached to: https://answers.gazebosim.org/question/22291/

oixel, 2025-02-23:
Another possible route to look at is the EntityComponentManager's "Each" functionality. If the position of each item can be grabbed through it, then the active sonar can be replicated by distance and angle calculations.

ElijaRock, 2025-02-26:
Updated the code to make the sonar joint spin so it can funciton as a sonar beam and collect data at multiple degrees of rotation. Right now it spins in the wrong axis, and trying to use the "axis" tag to change the axis of revolution causes the whole robot to spin even if it is only under the sonar joint.

Not really sure what is causing this or how to use a different tag if needed to change the axis. Maybe changing the initial pose or another tag could fix this. Also we still need to find out how to collect the data from this joint.

oixel, 2025-02-27:
This is an outdated plugin, but the functionality might be useful for referencing how to detect items in a given area.

https://classic.gazebosim.org/tutorials?tut=contain_plugin&cat=plugins

oixel, 2025-02-28:
Currently trying to repurpose Gazebo's pre-existing lidar sensor to serve as an active sonar. Having trouble getting the sensor to publish to a gazebo topic. This is the xacro currently:

```
<?xml version="1.0"?>
<robot xmlns:xacro="http://ros.org/wiki/xacro">
  <!-- Macro for inserting a Active Sonar -->
  <xacro:macro
    name="mil_active_sonar"
    params="name parent xyz rpy"
  >
    <!-- Active Sonar Link -->
     <link name="${name}_link">
      <origin xyz="0 0 -0.06" rpy="0 0 0"/>
      <inertial>
        <origin xyz="0 0 -0.06" rpy="0 0 0"/>
        <mass value="0.1"/> 
        <inertia ixx="0.001" iyy="0.001" izz="0.001" ixy="0" ixz="0" iyz="0"/>
      </inertial>
    </link>

    <!-- Active Sonar Joint -->
    <joint name="${name}_joint" type="fixed">
      <origin xyz="${xyz}" rpy="${rpy}"/>
      <parent link="${parent}"/>
      <child link="${name}_link"/>
    </joint>

    <!-- This ensures that the fixed link is not simplified away -->
    <gazebo reference="${name}_joint">
      <disableFixedJointLumping>true</disableFixedJointLumping> 
      <preserveFixedJoint>true</preserveFixedJoint>
    </gazebo>
    
    <!-- Sensor/Plugin Info-->
    <gazebo reference="${name}_link">
      <sensor
        name="Ping360" type="lidar">
        <pose>0 0 0 0 0 0</pose>
        <always_on>1</always_on>
        <update_rate>8</update_rate>
        <topic>/active_sonar/raw_data</topic>
          <lidar>
            <scan>
              <horizontal>
                <samples>720</samples>
                <resolution>1</resolution>
                <min_angle>-3.14</min_angle>
                <max_angle>3.14</max_angle>
              </horizontal>
            </scan>
            <range>
              <min>0.1</min>
              <max>30.0</max>
              <resolution>0.01</resolution>
            </range>
          </lidar>
        <plugin
          filename="ActiveSonar"
          name="active_sonar::ActiveSonar">
        </plugin>
      </sensor>
    </gazebo>
  </xacro:macro>
</robot>
```

A source that I referenced that might continue to be helpful is [Gazebo's example project using the Lidar.](https://gazebosim.org/docs/latest/sensors/)

oixel, 2025-03-07:
After looking at the .sdf and .urdf that get created on launch, the lidar sensor is being added properly to the world file; however, it still fails to show up in the gazebo topics list. I think the issue might be that I am trying to use a custom plugin instead of the gz::sim plugin.

oixel, 2025-03-07:
The most recent push in the single_beam_active_sonar has a line commented out in the ActiveSonar.cc:

`gz::transport::waitForShutdown();`

Removing the comment makes the active sonar's topic show up in gazebo's topic list, but prevents gazebo from loading.

oixel, 2025-03-12:
My most recent push finally got the active sonar to subscribe to its topic of "/active_sonar/raw_data". This means we can now move forward with making sure the right data is being published to the topic. Currently, basic test data of the sonar's twist is being fed through.

oixel, 2025-03-14:
Here is another good reference for sonar, lidar, and other sensor stuff:  https://github.com/MOGI-ROS/Week-5-6-Gazebo-sensors

oixel, 2025-03-14:
Update on status of lidar, regular lidar did not work, so active sonar is currently being implemented using GPU Lidar.

oixel, 2025-03-14:
GPU Lidar is set up, now a bridge needs to be built with the message from this repo:

https://github.com/CentraleNantesRobotics/ping360_sonar

oixel, 2025-03-14:
[image]

Active sonar is alive!

oixel, 2025-03-28:
This week we have been working more on implementing the bridge for the active sonar. I implemented the custom EchoIntensities message and have been working to pass the data from the gazebo sensors to ROS with Anthony.

oixel, 2025-03-28:
LaserScan is on the way to being converted from Gazebo to ROS2. Meanwhile, the data from PointCloudPacked seems to encrypted or corrupted? Will have to look into how to access the actual data from the PointCloudPacked message.

[image]
_Image of encrypted/corrupted data_

CarterKreis, 2025-03-30:
I helped Anthony and Oixel with some of their Gazebo to ROS bridge code.
- Finished bridge code for Gazebo LaserScan to ROS LaserScan
- Confirmed ROS node and Gazebo topic are properly obtaining data from simulation (for only laserscan)
- Need to finish code for Gazebo PointCloudPacked to ROS Sonar Echo bridge
- PointCloudPacked outputs proper data from gazebo, but needs to be written into custom ROS msg

oixel, 2025-04-04:
This issue is currently being put on pause until the Ping360 is actually purchased in order to ensure that the intensity data is accurate to the active sonar.

The current state of the issue is that although a lidar-based implementation was started, the PointCloud-Packed data might not be conveniently converted to the data type of int[] for intensities. Depending on what information is actually necessary from the Ping360, the scripts might have to be completely repurposed.

In the case that the only necessary information is the location of objects compared to Sub9 relative to a certain range, a potential execution would be to loop through all the objects in the scene using Gazebo's built-in Each function. The hydrophone plugin can be referenced to see how the functionality is used.