uf-mil/mil2 #30: Migration: `subjugator_keyboard_control`

cbrxyz, 2025-01-30:


TheRickyZhang, 2025-02-10:
Performed some initial migration, but was not able to test its functionality on Ubuntu yet. Largely focused on getting Ubuntu environment (dual boot, keyboard support/bindings) set up and reading through initial Jazzy documentation. Tried native Windows support but failed.

alexoj46, 2025-02-20:
@TheRickyZhang 

From meeting today:
- make copy of teleop package (done)
- modify copy
  - rename to subjugator_keyboard_control
  - publish geometry_msgs/Wrench instead of Twist

when that is done! (lmk when ready and I will give more specifics)
- remap keyboard to make sense for a sub in 6 degrees of freedom (x,y,z,roll,pitch,yaw)
  - ex: (up arrow = +x, down arrow = -x, etc.)

alexoj46, 2025-04-10:
Update on this issue: the package has been migrated successfully, just needs a bug fix where pressing and holding a key works for a moment, then goes to zero output, then back to desired output, instead of having a steady correct output when holding a key down.

commit aa470a84 (2025-05-20): Updated keyboard control logic (#147)

* Updated keyboard control logic

* pre-commmit fixes

closes #30

---------