uf-mil/mil2 #31: Migration: `dvl-a50`

cbrxyz, 2025-01-30:


alexoj46, 2025-02-04:
Old ROS 1 driver submodule here: https://github.com/uf-mil/dvl-a50-ros-driver

sgupta1703, 2025-02-09:
# Weekly Report

## Summary
This week, I focused on setting up my development environment and conducting research related to MIL (Marine Intelligent Laboratory) tasks. Below is a breakdown of my progress.

## Tasks Completed

### 1. Ubuntu Installation on Virtual Machine
- Successfully installed Ubuntu on a Virtual Machine to create a suitable environment for ROS2 and MIL-related development.
- Ensured system settings were optimized for smooth performance.

### 2. Installed Required Packages
- Installed all necessary dependencies and packages required to get started on MIL.
- Verified the installation of ROS2 and relevant libraries.

### 3. Research on DVL Driver for ROS2
- Read through my assigned issue: **Find a good DVL driver for ROS2**.
- Researched what a **Doppler Velocity Log (DVL) driver** is and its role in underwater navigation.
- Understood how DVL data is published in ROS2 and the key features to look for in a suitable driver.

### 4. Found Relevant Repositories
- Identified and reviewed multiple repositories offering DVL drivers for ROS2.
- One promising repository I found:
  - **[Waterlinked DVL Driver](https://github.com/Robotic-Decision-Making-Lab/waterlinked_dvl)**
  - Provides a ROS2-compatible interface for Water Linked DVL devices.
  - Includes features for velocity estimation and underwater navigation.

## Next Steps
- Evaluate and test the identified DVL driver in a ROS2 simulation.
- Compare different DVL drivers to determine the best fit for MIL’s requirements.
- Begin integration and testing with ROS2 nodes.

---
**End of Report**

alexoj46, 2025-02-13:
Forked your first link for adding as a submodule here! https://github.com/uf-mil/waterlinked_dvl

alexoj46, 2025-02-20:
Can use this command to install the necessary marine_acoustic_msg package: `sudo apt install ros-jazzy-marine-acoustic-msgs`

sgupta1703, 2025-02-23:
```markdown
# MIL Weekly Report

## Week Summary

### Tasks Completed
- Checked out the [waterlinked_dvl](https://github.com/Robotic-Decision-Making-Lab/waterlinked_dvl) repository.
- Encountered issues with missing configuration packages during installation.
- Resolved missing package issue by:
  - Using `vcs` to import `src/waterlinked_dvl/ros2.repos`.
  - Running:
    ```sh
    rosdep install --from-paths src -y --ignore-src --skip-keys nlohmann_json
    ```
    to install missing dependencies.

### Ongoing Work
- Working on installing `marine_acoustic_msg_package` using ROS Jazzy.

### Challenges
- Initial setup issues with missing dependencies, which were resolved through `rosdep` installation.

### Next Steps
- Complete the installation of `marine_acoustic_msg_package`.
- Test and validate the functionality of the `waterlinked_dvl` package in the ROS Jazzy environment.
```

sgupta1703, 2025-02-23:
# MIL Weekly Report

## Week Summary

### Tasks Completed
- Checked out the [waterlinked_dvl](https://github.com/Robotic-Decision-Making-Lab/waterlinked_dvl) repository.
- Encountered issues with missing configuration packages during installation.
- Resolved missing package issue by:
  - Using `vcs` to import `src/waterlinked_dvl/ros2.repos`.
  - Running:
    ```sh
    rosdep install --from-paths src -y --ignore-src --skip-keys nlohmann_json
    ```
    to install missing dependencies.

### Ongoing Work
- Working on installing `marine_acoustic_msg_package` using ROS Jazzy.

### Challenges
- Initial setup issues with missing dependencies, which were resolved through `rosdep` installation.

### Next Steps
- Complete the installation of `marine_acoustic_msg_package`.
- Test and validate the functionality of the `waterlinked_dvl` package in the ROS Jazzy environment.

sgupta1703, 2025-02-23:
quick note to install the following packages:

`sudo apt install ros-jazzy-marine-acoustic-msgs`
`sudo apt install ros-jazzy-generate-parameter-library`

sgupta1703, 2025-03-02:
# Weekly Report

-Finished building the driver with success
-Problems arise in running the actual driver as the process dies immediately on run.
-Tested out the driver's IP address

## Ongoing

-Fixing the problem with the process just ending after starting.
-Not entirely sure why its doing that. Have to look into it.

alexoj46, 2025-03-06:
@sgupta1703 I took another look at the driver and found the issue we were having. Turns out the actual yaml file used for configuration was 'waterlinked_dvl_driver_parameters.yaml' inside the src folder, not just the one in config. Changing the IP address there fixed the problem. Great work finding this driver and getting it to run! I will merge it in later today!

alexoj46, 2025-03-06:
Closed in 7c3dca2

closed 2025-03-06 by alexoj46