uf-mil/mil2 #41: Migration: `ros_alarms`

cbrxyz, 2025-01-30:


tjcanro, 2025-02-10:
# Migration Plan

## Required Steps

- [ ] Update setup.py to new format in the ROS2 environment
- [ ] Update launch file to use python instead of XML format
- [ ] Since the package makes use of axros (Designed for ROS1), update to use the ROS2 built-in asynchronous package
- [ ] Re-write use of rospy to the new rclpy used in ROS2 for main init and alarms.py code
- [ ] Migrate the alarm server, clear, report and raise nodes
- [ ] Ensure compatibility within all updated nodes
- [ ] Update tests to rclpy and rclcpp, as well as research and implement ros2 testing

## Testing

- Implement python and c++ test with alarm broadcaster and listener, asserting the periods in which the alarm should be raised and cleared
- Callback testing
- Heartbeat testing update
- Replace axros tests for async ros2 package tests

## Definition of done

- All nodes and main package functionality updated to use rclpy
- Setup and initialization of package done using colcon build
- Comprehensive tests for all nodes passing

## Questions

- Why do we have tests implemented in c++ and separate python tests for this python package? Is there any way in which we could have comprehensive testing in one language?
- How time sensitive is this package? Would it make sense to migrate it to c++ for the performance improvement?
- Need more information on the workflow of our tests. How do we integrate python packages in c++ tests?
- Need explanation on use of test_handler.py
- Should we add the following to our setup.py to have automatic tests?:
``` tests_require=['pytest'] ```

DaniParr, 2025-02-10:
### Answers
- Write tests only in python
- Keep it in python for now
- Keeping it python
- Ignore this file if it is not imported from anywhere
- Yeah

tjcanro, 2025-02-17:
## Progress Update

This week, I have created the main components of the package. I have fully migrated the ros_alarm_msgs package and I am currently working on updating the rest of the package. Throughout my process for migrating ros_alarms, I have had to do a lot of research on the differing functionalities between ros2 and ros1 packages. Specifically, there has been a lot to learn on the configuration files, such as the setup.py

## New Questions

- I was under the impression this was a pure python package. However, I noticed there is an alarm proxy and a broadcaster in C++. Does this mean this is a hybrid package? If so, do we want it to remain that way?
- I might need some more explanation on how to use only the ros2 asynchronous python functionality instead of axros

tjcanro, 2025-03-03:
# Progress Update

This week, I managed to re-write the header files for the different nodes that belong to the ROS alarms package. This package has more functionality implemented than I initially approximated, so I have had to do a lot of research on changes for the migration to be effective. I also worked on the setup of this hybrid package. Although I am waiting to have some more functionality ready, I have implemented a near-final version of the CMakeLists, the package.xml and the setup.py. For now, I believe there is a lot of good progress in this package, and its integration with the ros2_alarm_msgs package.

tjcanro, 2025-03-31:
## Weekly Update

This week, I focused on testing. I set up the testing framework on the CMake and I verified linking with the package. Additionally, I managed to complete and fully test the alarm proxy. I am now working on the tests for the other sections of the package that I have written. Now that the setup is complete, I will focus on finishing tests and migrating the remaining sections. I have also added a scripts file with an alias for easier and more visual testing

## Questions

For now, I don't have any pressing questions, I will most likely need some help migrating the asynchronous parts of it

tjcanro, 2025-04-07:
# Weekly Update

This week I focused on the implementation of asynchronous spinning for this node. After doing some research on how this is supposed to operate on ros2, I have decided to use a callback group combination with multi-threaded executors. 

Because asynchronous processes and parallelism are more hands-on in ROS2, I am leaving some context for how I am refactoring this section. Ros2 handles parallel callbacks in the following way:

### Asynchronous Nodes

In order to spin alarms asynchronously, this is the approach I am taking:

- Use of a multi-threaded executor: `rclcpp::executors::MultiThreadedExecutor` allows multiple callbacks to be handled in parallel. This is critical in our case because alarm handling should not block other callbacks or the main thread
- Create dedicated callback group: The subscription for alarms is placed in a CallbackGroup of type Reentrant, which means multiple callbacks associated with the same group can be executed concurrently

### References

To better understand I referenced the following posts and documentation:

- https://discourse.ros.org/t/how-to-use-callback-groups-in-ros2/25255
- https://docs.ros.org/en/foxy/How-To-Guides/Using-callback-groups.html
- https://docs.ros.org/en/foxy/Concepts/About-Executors.html#executors

tjcanro, 2025-04-21:
## Weekly Update

This week I have focused on writing the listener node. I managed to implement a multi threaded executor that can handle asynchronous calls. In order to avoid deadlocks, I decided to use a Mutually Exclusive Callback Group. From my understanding, this will still allow parallel calls as long as they are coming from different callback groups, while allowing every alarm to complete. At the moment, I am developing the tests for the listener, as I have most of the functionality.

- Additionally, now that the listener is almost complete, I can test the broadcaster as well, which was also complete.

### Next steps

- Now, I will focus in finishing tests and preparing a PR for the C++ portion of the node, as it is such a big package and I would like to have the python implementation in a different PR