uf-mil/mil2 #50: Create launch file for mil_passive_sonar

alexoj46, 2025-02-04:
We want to be able to launch the sdgps pipeline and ping publisher node from a single launch file.

tripAarush, 2025-02-24:
Started on this issue, made an outline of things to learn and implement.
did:
set up the ros file for sdgps_pipeline and ping_publisher
Confirmed the package names
Checked if any parameters need to be passed to the nodes
to do:
make the function for the launch file and test it
Commit the file into the ros2 package

alexoj46, 2025-02-24:
Hey @tripAarush! For now, focus your efforts on issue #51, and if you are looking for more after making a PR for that let me know! To actually test this issue, you'd need to install a third party library (which an alumni would need to grant permissions for) so it is something I assigned to myself.