uf-mil/mil2 #51: Setup joystick control for sub9

alexoj46, 2025-02-07:
Goal is to get input from an Xbox controller and publish it to a ROS topic, so that we could control sub9 with the controller.

Look into this package: https://index.ros.org/p/joy/

tripAarush, 2025-02-10:
I met up with our team this week and installed and fixed my ros2 and ubuntu. I created a node on my ros2 workspace and added C++ dependency with VS code. I am having issues with the configuration and importing ros into vscode, but i have the code to get the directions for an input, and I am going to test it with the controller this week when I get my code to run.

tripAarush, 2025-02-17:
Finished up the code for getting directions from the controlled. It works and gives directions for the controller I used to test, and I will double check that it works with the actual controller as well.

alexoj46, 2025-02-26:
@tripAarush From meeting today:
- Make a folder called "command" inside the subjugator folder, add your new package here (call it "subjugator_joy") here
- Your node should:
  - Subscribe to the joy topic
  - Use the information from the joy msgs to publish a new [geometry_msgs/Wrench](https://docs.ros.org/en/noetic/api/geometry_msgs/html/msg/Wrench.html) msg to a topic called "/cmd_wrench" whenever you hear a new joy msg
    - for example, might publish a wrench with a +1 Y force when the left joystick is pushed fully forward"
    - map it however makes sense to you for the 6 axes, and we can change it later if need be

tripAarush, 2025-03-02:
Finished the file and published the joy node into wrench. I learned Xbox controller is conventionally -1 when moving the joystick to the right and +1 for the left. Will talk about what we prefer and edit it based on that as well.

PR #77 (2025-03-12): Getting joystick info
Fixes #51
closed 2025-03-12 by alexoj46