uf-mil/mil2 #52: Create C++ `enumerate` functions

jojongx, 2025-02-07:
We need a C++ template function that features the equivalent of Python's `enumerate`. This should be placed in `mil_tools::itertools`.

andrew-aj, 2025-02-07:
You could implement something like 
[this](https://www.reedbeta.com/blog/python-like-enumerate-in-cpp17/). It allows you to get access to a tuple that you can unpack and refer to the elements with their own variable names like in python.

jojongx, 2025-02-09:
Does something of this format need to be added to the CMakeLists.txt?

ament_add_gtest(${PROJECT_NAME}_string_test test/string.cpp)
  target_include_directories(${PROJECT_NAME}_string_test PUBLIC
    $<BUILD_INTERFACE:${CMAKE_CURRENT_SOURCE_DIR}/include>
    $<INSTALL_INTERFACE:include>
  )
  target_link_libraries(${PROJECT_NAME}_string_test ${PROJECT_NAME})

I just took this snippet from the file for an example.

cbrxyz, 2025-02-09:
You will need to modify `CMakeLists.txt` in some way, how you modify it will depend on your implementation. You probably will need to add a new `ament_add_gtest` call for (presumably) your new file!

jojongx, 2025-02-10:
This week, I met with Cameron and decided to contribute to the mil2 repo's helper functions. I was tasked with writing an enumerate function for C++, mimicking the same function found in Python. I spent my time refreshing myself on template functions, learning constexpr, and exposing myself to the repo's src folder. @andrew-aj shared a [resource](https://www.reedbeta.com/blog/python-like-enumerate-in-cpp17/) that provided the helper template function we needed, so I made the necessary files (to match the organization/layout of similar files) and implemented this code. I made a pull request and need to make amends to my testing, CMakeLists, etc. Currently, I am writing my tests and trying to figure out the cause of my build errors. I will continue working on, and hope to finish, the issue this upcoming week.

closed 2025-03-16 by jojongx