uf-mil/mil2 #53: Gazebo: Create torpedo launcher actuator

alexoj46, 2025-02-07:
Need to create a simulated torpedo launcher in the Gazebo simulation for sub9.

sheff2, 2025-02-10:
-setup VM, ros2, and gazebo
-watched whole tutorial series on node, publisher, subscriber, topic, and services
-currently working on actually creating torpedo launcher actuator

sheff2, 2025-02-16:
-learned how to make .sdf files and make my own robot
-starting testing things in mil2 gazebo (published floats to thrusters to make the sub move)
-trying to figure out the best way to connect the torpedo to the sub

nicholasfarr, 2025-02-17:
- familiarized myself with .sdf files
- worked with Shaun to brainstorm how to connect the torpedo to the sub
- fully switched from UTM to parralels.
- Going to start implementing solutions next week -> getting up to speed and familiar with codebase.

sheff2, 2025-02-24:
-found torpedo documentation from Daves robots
-will try to implement that this week

nicholasfarr, 2025-02-24:
- Reviewed the torpedo documentation found by Shaun from Dave's robots.
- Continuing to familiarize myself with the existing codebase and actuator implementations.
- Planning to collaborate with Shaun this week to implement and test actuator solutions for integrating the torpedo into the sub.
- Will update progress as implementation proceeds.

sheff2, 2025-02-27:
-Going to add either the dave robot torpedo model or the 2024 one thats currently in this repo
-going to test adding force to it underwater and see how it acts
-then i will turn it into a script to have it launched programmatically

sheff2, 2025-03-09:
-added the old torpedo to the main world
-created a config file for the gazebo
-using the add force tool works as expected on the torpedo
-now i need to make a script that unlinks it from torpedo and adds a force to simulate a launch

sheff2, 2025-03-17:
-made a script to publish force values to simulate torpedo being launched
-working on trying to have the add force plugin subscribe to it so I don't have to manually enter in forces

sheff2, 2025-03-31:
using a ros2 launch script to spawn in torpedos into running simulation
-need to make it add force
-need to fix collisions (torpedo falling through ground)
-need to figure out how many torpedos to add and what parts of the sub to add it to

sheff2, 2025-04-07:
Working on c++ plugin that attaches to sub and then spawns torpedos using entity spawner at the pose of the sub. No longer using the python spawn file because I can't get the subs pose that way.