uf-mil/mil2 #54: Gazebo: Create a claw in gazebo

DaniParr, 2025-02-07:
- [ ] Find model of the claw
- [ ] Figure out plugins on how to move components
- [ ] Prove in simulation that the claw works
- [ ] Attach the claw to sub
- [ ] Have the claw pick up items and interact with environment

heirynhr, 2025-02-10:
- Set up VM
- Studied up on Ros 2 and gazebo
- currently looking for claw models

DaniParr, 2025-02-10:
- [ ] Create your own project
- [ ] Have the claw model display on gazebo
- [ ] Create a simple publisher that takes in keyboard controls to either open or close the claw

jojongx, 2025-03-16:
Per the guidance of @DaniParr and @Keith-Khadar , I worked through the Gazebo plugin tutorials and learned how to make a plugin between studying for midterms. I am going to look into the example plugin next and get more familiar with the Gazebo environment

jojongx, 2025-03-30:
I spoke with @Keith-Khadar for a project walk-through.  The intended workflow is generally xacro -> gazebo config/placement -> plugin. I created a basic claw.xacro with positional information (xyz and rpy similar to other xacros) and added this file to the main sub9 xacro. Still need to finish the file — any info regarding how the thruster.xacro was made would be useful!

jojongx, 2025-04-07:
I could not get in much time this week due to other work.

jojongx, 2025-04-13:
I added the link and joint for the claw to the urdf file. I will leave it as is for now and will circle back to it when I get more work done in Gazebo/plugin (so I better understand what is necessary etc.). I created the claw plugin header and source files with basic implementation using the example plugin. I got more familiar with Gazebo and (hopefully) will be able to leverage plugin work with the simulation to make meaningful progress+better my understanding.