uf-mil/mil2 #57: Revamping Underwater camera plugin

TaherA51, 2025-02-10:
Me and Charlie are assigned to revamp the underwater camera plugin from Project Dave which has limited simulation ability and features.

cdurkin35, 2025-02-10:
- Setup VM and mil2 repository
- Reviewed and learned ROS2 and Gazebo
- Read the documentation for the Underwater Camera Plugin on Project Dave
- Discussed with Taher plans on the design of the plugin and timeline

cdurkin35, 2025-02-17:
- Learned more about ROS2
- Went in depth on documentation of Gazebo, specifically on creating plugins
- Reviewed source code of Project Dave's plugins
- Researched methods to implement the 'turbidity' parameter, and started to write an outline for it

TaherA51, 2025-02-17:
Worked independently, and with Charles on similiar tasks as him. 

- Documentation of gazebo
- Project Dave plugin source code review
- Brainstorming ways to implement turbidity parameter for the underwater camera plugin.

cdurkin35, 2025-02-24:
- Continued research in developing the turbidity feature of the plugin
- Began code development to implement this feature
- Looked into ways of implementing UI for modifying parameters in the plugin and drafted ideas for design
- Reviewed more ROS2 and Project Dave documentation
- Planning on meeting with Taher at some point this week to go over progress and make plans for further development

TaherA51, 2025-03-02:
This week I spent some time reading through the prexisting code base for the Underwater Camera plugin and the other plugins to understand how plugin development in Gazebo work. Using this foundation, I'm planning to add a new turbidity element into the SDF file so users can set this parameter. This turbidity element would be stored as a private float member variable in the class. The next change I think we can make is a 'noise' layer that can be added after the per-pixel attentuation loop; after researching a bit about CV, I think we could use the randu to generate this noise layer that could be used to simulate plankton or detritus. The previous turbidity parameter will be used to adjust density and the size of the particles. 

Additionally, it would be really helpful to see real underwater footage from our bot from previous competitions. Seeing the actual footage will provide a clear benchmark and help to understand which features we need to better match the real-world environment.

cdurkin35, 2025-03-03:
- Completed turbidity feature of underwater camera plugin (Having problems with git, will push code in the next few days)
- Debugged errors with install script for zsh and opened issue for it
- Talked to Keith about future goals for underwater camera plugin, and began researching methods to implement
- Allowed more customizable options in turbidity parameter

cdurkin35, 2025-03-10:
- Added more options in the SDF file for the turbidity plugin that allow for the customization of particle color, particle size, and weight of overlay
- I am still having problems where I cannot create a PR for this code, so I will go to the lab this week and get that resolved to get turbidity working in our future simulations
- Began researching UI designs to update the parameters while the simulation occurs

TaherA51, 2025-03-10:
Had a busy week with midterms and other commitments, so I couldn’t make much progress—looking forward to getting back on track soon.

cdurkin35, 2025-04-07:
Have had a busy last few weeks so apologies for the slow progress on this plugin, however I was able to get some work done this week.

- Added finishing touches to turbidity feature to allow for several parameters in SDF file to control the behavior of the plugin
- Began work on 'volumetric rays' feature. There is some documentation of this on the old Project DAVE repo, so I have been using this as a base.
- Began work on fixing ZSH compatibility for shell scripts as mentioned in issue #73 
- Will continue work on the plugin, and be in the lab this week to create a PR with fully functional turbidity feature and basics of 'volumetric rays' feature.

TaherA51, 2025-04-07:
Been in contact with Keith. Progress on my end has been slow, and I am taking a pause for now due to underestimated courseload.

cdurkin35, 2025-04-14:
Continued work on 'volumetric rays feature', more work than anticipated, will take a little longer to fully implement and debug

cdurkin35, 2025-04-21:
Continued work on volumetric rays feature.