uf-mil/mil2 #60: Gazebo: Hydrophones and Pingers

CarterKreis, 2025-02-14:
Set up hyrdophones on sub and Pingers in course all in Gazebo

CarterKreis, 2025-02-14:
- Will create plugin from scratch for Hydrophone to gather information about Pinger and find its direction 
- Check ExamplePlugin.cc in Repo for help
- Create Hydrophones Models on sub

lucasperez-techdev, 2025-02-17:
This week I sadly wasn't able to come into the lab but I was still able to continue learning about ROS and Gazebo. I spent my free time watching Kevin Robotics videos to familiarize myself more with working with gazebo and ROS. On friday I spoke with Carter about working on the new Hydrophones plugin issue. Over the weekend I spent time trying to familiarize myself with creating custom plugins from scratch and how hydrophones worked. This coming week Ill be in the lab trying to collaborate on this issue further.

CarterKreis, 2025-02-23:
- Filled out Hydrophone.cc/hh but does not run
- Will look into how to iterate through every Gazebo component using Each() call
- Created ROS2 message ProcessedPing.msg
- Added Hydrophone.cc/hh and ProcessedPing.msg to Packages.xml and CMake for relative directories

CarterKreis, 2025-02-28:
- Learned more how the Each() call works
- The plugin is running and publishing ros messages
- The plugin is iterating through components in the Sub9 SDF rather than the World SDF
- By next week, the Plugin should be running and correctly pulling from the World SDF

lucasperez-techdev, 2025-03-03:
- Fixed the orientation of the hydrophone on the sub9 model
- Worked with Carter on testing the functionality of the custom plugin
- Talked with Kieth about next steps after the hydrophone plugin is finished and integrating files for the MIL rover into mil2 repo
- For the next week, I am going to continue working on testing the functionality of the hydrophones
- Planning on starting to create the world for the rover that is based off of the third floor of malachowsky where the rover will be mainly tested

CarterKreis, 2025-03-10:
Busier week with schoolwork, so not as much progress as I would have hoped
- Figured out a way for Plugin to iterate through the World SDF and not the Sub9 SDF
- Confirmed that Plugin successfully grabs Pose for an arbitrary number of Pingers
- Pinger Frequencies are still not being pulled from World SDF
- Sub9 information looks like it can be grabbed from World SDF info so will have to figure that out to get Sub Pose

I am hoping to be testing the Plugin by the end of this week

CarterKreis, 2025-03-17:
Pull request was made for this issue
- Pinger Frequencies are getting pulled from WorldSDF by using Pinger Names
- Sub9 Position is updated on post-update() and difference vectors between pingers and sub9 are calculated
- ROS2 msg is properly being sent and received

closed 2025-03-17 by CarterKreis