uf-mil/mil2 #62: Create Gazebo auto-scoring RoboSub plugin

cbrxyz, 2025-02-16:
We should make a Gazebo plugin that automatically scores a RoboSub run. This could be done by observing when the robot does actions that earn points, such as:

1. Successfully navigating through a channel
2. Touches a certain object
3. Surfaces in the correct area

For the exact list of requirements, consult the rulebook (can be found on the wiki/robosub website).

Carlosdc25, 2025-03-03:
This week I looked into tasks sub9 would do for the RoboSub competition, and created a ScorePlugin.cc and .hh file to check when the sub arrived/passed a gate. I also made changes to CMakeList.txt and sub9.urdf.xacro to ensure the plugin can be seen. I am publishing the current score when the task is complete to a

Carlosdc25, 2025-03-10:
This week I mostly worked on assigning fewer points when the sub passes through a gate at a non-direct angle. To do this I got the local forward direction of the sub and the gate we are passing through. You can't simply compare these local forward directions, you must first transform the vectors into world coordinates using the RotateVector() function . This was important since the local forward directions of the models themselves are only relative to themselves so they cannot be used for comparison. Then using the new vectors i got, i plugged them into a cosine formula and then got the angle between the gate and sub. Haven't tested this yet. I am worried that I am not passing in the right local forward direction.

Carlosdc25, 2025-03-31:
This week I made steps to testing the points the sub would get from rolling, pitching and yawing through the gate. I increased the angular speed from the subjugator keyboard controls and now I am getting the sub to be able to identify when it has spun and passed through a gate simultaneously. I am doing this by getting the degrees rotated and then having that each set of 90 degrees equals 1 band(rounding down), the amount of bands will dictate the points.

Carlosdc25, 2025-04-07:
This week I worked on making a node that rolls, pitches, or yaws the sub to more accurately test the sub's scoring when it came to these metrics. The purpose is to make the tests more reproducible. I am also working on spreading the plugin's functionality into multiple files to be more organized and readable.

Carlosdc25, 2025-04-21:
This week I fixed an issue I have been previously having with recognizing when the sub passed through a gate, before I made a radius around the center of the gate and if the sub entered that sphere then it would be considered a pass for the scoring gate. But now I configured the size of the gate passing zone to be a rectangular thin cube shape that better represents the sub passing through the gate. Additionally, this space that makes the sub consider that it has passed a gate is relative to the gate. In other words the dimensions of the gate's passing space will always be relative to the gates position in the world. This way issues dont come up when the sub is at a different angle from a gate.