uf-mil/mil2 #63: Re-write preflight in rclcpp

cbrxyz, 2025-02-16:
`preflight` has been a trusty tool that our team has consistently used to ensure our vehicles are ready for deployment. However, we should try to re-write preflight in C++ for ROS2 support, performance reasons, and to fix a few long-running issues (namely, excessive warnings/errors from some python async functions).

zrzz-hq, 2025-02-24:
Here is a detailed plan the new C++ version preflight

[pre-flight_plan.pptx](https://github.com/user-attachments/files/18947211/pre-flight_plan.pptx)

zrzz-hq, 2025-03-03:
I've spent some time getting familiar with the FTXUI library, which differs from traditional GUI frameworks like Qt. I designed the first version of the UI but found the code structure lacking, so I rewrote it.
**Current UI Implementation**:
- Main Menu
- Test Page
**Planned UI Features**:
- Read Documentation Page (Requires parsing the README file)
- View Report Page
Once the UI is complete, I'll move on to implementing plugins.

zrzz-hq, 2025-03-10:
**Current UI Implementation:**
- Main Menu
- Test Page
- Report Page

**Current Plugins Implementation.**
- Node plugin
- Topic plugin
- Setup plugin
- Actuator plugin

**Known issue**
- Cannot exit if dialog is not closed
- Cannot display multiline text in the dialog and report page

closed 2025-03-28 by DaniParr