uf-mil/mil2 #71: Gazebo: Create rover world

cbrxyz, 2025-02-23:
We will need to make a basic world for the rover. The complexity of the world can be up to the person who chooses to take on this task, but I think we should start with something simple, and then work our way up to more complex objects/environments.

We could:
* Build a world with only the default objects included in Gazebo
* Attempt to recreate the third floor of Malachowsky (where a lot of the rover will probably be tested)
* Build another part of campus, such as outdoors, MAE-C, or MAE-B

lucasperez-techdev, 2025-03-09:
- Started working on the Gazebo Simulation files for the Rover in the reptile rover branch
- Researched resources for creating the world file and familiarized myself with the process
- Talked with Joseph Goodman about what type of world is needed, decided on a relatively simple flat world with small bumps/textured terrain
- Planning on having the simulation running by the end of this week so that plugins and rover functionality can be tested

lucasperez-techdev, 2025-03-17:
This past week I wasn't able to come into the lab much for help on this issue due to midterms/exams and whatnot but I am still working on getting the simulation to work and figuring out what exactly is needed for the rover files. Ran into some issues creating the cmake files for each folder/linking everything. Hopefully after spring break I can come into the lab and figure this out with leadership and further discuss what is needed in the rover sim.

lucasperez-techdev, 2025-03-31:
Getting back into the groove of things, I was out of town most of this past week and wasn't able to access my computer but the last couple of days I was debugging some of the cmake linking issues with the rover bringup and rover launch files I made. Going to be in lab more this week to talk to Keith about it to hopefully be able to have this simulation functioning soon.

lucasperez-techdev, 2025-04-06:
Finished the setup of the files for Rover in MIL2. Very basic world is setup and able to be launched. This week I am going to work on building the world with obstacles for testing and then trying to model the rover in Gazebo.

lucasperez-techdev, 2025-04-14:
Wasnt able to work on designing the rover world this past week because I was traveling but this week I will finish designing and get in touch with Joseph to make sure the necessary obstacles are included in the world