uf-mil/mil2 #79: Fix colbuild symlink error

Keith-Khadar, 2025-03-07:
This error seems to pop up after running colcon build then  colbuild.

[image]

Keith-Khadar, 2025-03-07:
- [ ] Add another alias for colbuild that is just CB. 

- [ ] Also make it so you can colbuild for a specific package

alexoj46, 2025-03-13:
@sgupta1703 to replicate:
in mil2 folder:
rm -rf log install build
colcon build
colcon build --symlink-install

figure out why/how to fix

sgupta1703, 2025-03-15:
The error arises because colcon attempts to create a symbolic link at a specific location, but a directory already exists at that path. The build process cannot overwrite the existing directory with a symlink, leading to the failure.

Probably Because of these issues:

Leftover Files from Previous Builds: Residual directories or files from previous failed or incomplete builds may cause conflicts when colcon attempts to create symlinks.

Incorrect Permissions: If files or directories are owned by a different user (e.g., root), the current user may lack sufficient permissions to delete or modify them, resulting in a failure when attempting to create symlinks.

Symlink Conflicts: There might be situations where the symbolic link target is already a directory rather than a symlink, causing colcon to fail when trying to overwrite it.

Removing the Conflicting Directories: For each package that failed (e.g., mil_msgs, electrical_protocol), the conflicting directory was manually deleted:

rm -rf /home/MIL_USER/mil2/build/<package_name>/ament_cmake_python/<package_name>

sgupta1703, 2025-03-23:
These are the 4 directories that are giving problems:

MIL_USER@MIL:~/mil2$ rm -rf /home/MIL_USER/mil2/build/mil_msgs/ament_cmake_python/mil_msgs/mil_msgs

MIL_USER@MIL:~/mil2$ rm -rf /home/MIL_USER/mil2/build/mil_passive_sonar/ament_cmake_python/mil_passive_sonar/mil_passive_sonar

MIL_USER@MIL:~/mil2$ rm -rf /home/MIL_USER/mil2/build/subjugator_msgs/ament_cmake_python/subjugator_msgs/subjugator_msgs

MIL_USER@MIL:~/mil2$ rm -rf /home/MIL_USER/mil2/build/electrical_protocol/ament_cmake_python/electrical_protocol/electrical_protocol

*ignore the MIL_USER@MIL

-> removing these 4 directories allows for a clean colcon build --symlink-install after clearing logs/builds/install folders and doing a colcon build

sgupta1703, 2025-03-31:
Creating a new bash function is easiest because these directories:

MIL_USER@MIL:~/mil2$ rm -rf /home/MIL_USER/mil2/build/mil_msgs/ament_cmake_python/mil_msgs/mil_msgs

MIL_USER@MIL:~/mil2$ rm -rf /home/MIL_USER/mil2/build/mil_passive_sonar/ament_cmake_python/mil_passive_sonar/mil_passive_sonar

MIL_USER@MIL:~/mil2$ rm -rf /home/MIL_USER/mil2/build/subjugator_msgs/ament_cmake_python/subjugator_msgs/subjugator_msgs

MIL_USER@MIL:~/mil2$ rm -rf /home/MIL_USER/mil2/build/electrical_protocol/ament_cmake_python/electrical_protocol/electrical_protocol

--Are all in the ament_cmake_python file in their respective build folders. 

all files follow this format:

/home/MIL_USER/mil2/build/build_folder/ament_cmake_python/build_folder/build_folder

so easy to delete

sgupta1703, 2025-04-05:
rebuild() {
    local ws_root="$( cd "$( dirname "${BASH_SOURCE[0]}" )" && pwd )"
    cd "$ws_root"

    echo "Cleaning previous build artifacts..."
    rm -rf log install build

    echo "Building workspace..."
    colcon build

    echo "Removing specific auto-generated Python directories..."
    for pkg in mil_msgs mil_passive_sonar subjugator_msgs electrical_protocol; do
        rm -rf "build/${pkg}/ament_cmake_python/${pkg}/${pkg}"
    done

    echo "Rebuilding with symlink install..."
    colcon build --symlink-install

    echo "Workspace rebuild complete."
}

sgupta1703, 2025-04-21:
Finally,

I opened a pull request for this particular function as this completely cleans up any auto generated symlinked Python directories.

```

rm_symlink_dirs() {
    cd "$MIL_REPO" || return

    echo "Removing auto-generated symlinked Python directories..."
    find build -type d -path "*/ament_cmake_python/*/*" -exec rm -rf {} +

    echo "Done."
}
```

commit 59cdfb64 (2025-04-29): Fixes cb symlink build issue (#148)

closes #79

* fixed symlink issues

* updated path to not have to cd

* added to cb! (when building all packages)

* made more robust to missing dirs for CI

---------
closed 2025-04-29 by alexoj46