uf-mil/mil2 #81: `colbuild`: Add `--verbose`, and package limiting

cbrxyz, 2025-03-08:
We should add the following to `colbuild`:
* `--verbose`: `--event-handlers console_cohsion+ --cmake-args -DCMAKE_VERBOSE_MAKEFILE=ON`
    * Prints out all logs
* Positional parameters: Go into `--packages-select ...`
    * `colbuild electrical_protocol` only builds `electrical_protocol`

tripAarush, 2025-03-31:
Updates while working on this: I added support for building multiple specific packages by allowing the colbuild command to accept a variable number of arguments. I removed the automatic deletion of build, install, and log directories to avoid unnecessary recompilation. One part I still need to finish is making sure the verbose flags are only applied when verbose is explicitly passed, without affecting normal or multi-package builds.

alexoj46, 2025-04-05:
@tripAarush Earlier today, seems a related PR (https://github.com/uf-mil/mil2/commit/b2b98c5a2251d2f359f12e6cd69d9b7aefc139b4) was merged in that adds some autocomplete functionality, as well as the ability to specify which packages to build. If you could merge your solution for the verbose flags in with these new changes that would be very helpful!

closed 2025-04-15 by alexoj46