uf-mil/mil2 #82: Implement RL model for basic SubjuGator tasks

willzoo, 2025-03-09:
Continuation of issue linked at: https://github.com/uf-mil/mil/issues/1325

### **What needs to change?**

We are implementing a basic RL model algorithm to SubjuGator using the ROS2 simulation so that the robot can automatically learn how to perform basic tasks. We will be using the PPO algorithm outlined in the research paper https://arxiv.org/abs/1707.06347, and eventually writing our own research paper based on the results of this project.

### **How would this task be tested?**

1. Download Python dependencies
2. Run a stable_baselines3 script that connects to the ROS2 simulation
3. Analyze stats about training/evaluation

danushsingla, 2025-03-09:
Spoke with Will and Mohana about changing our plan a little. Instead of using the entire image matrix into the RL model (which training could be slow) we would calculate pose values of the buoy and send that to the RL model so it has less training work to do.

willzoo, 2025-03-10:
I met with Danush and Mohana this week. As Danush said, inputting the entire image into the RL model's observation space might be very computationally intensive, even with max pooling. The first task we want the RL agent to complete is to train itself to collide with a buoy in front of it, and if we continuously input the buoy's pose relative to the sub, it should provide enough data for this. Additionally, I have created a new branch for this issue, and created a new gazebo world, for us to make custom testing environments for the RL agent. Currently the world is just the sub and a buoy. Next week, we need to start building the gym environment itself, and to find a way to programmatically receive the pose of the buoy from the sim.

mohana-pamidi, 2025-03-10:
This week, we will work on getting data from the perception module to detect the buoy and it's pose relative to the sub. This information will be sent over to our RL model.

danushsingla, 2025-03-16:
Was unable to work on this issue this week due to mid-terms.

mohana-pamidi, 2025-03-17:
This week, I was looking more into how exactly we will be creating our state and observation spaces for

danushsingla, 2025-03-30:
Met with Will to create the rest of the publishers and subscribers needed to get information from the submarine. Used some of Mohana's code as well to create the action spaces for the RL model. Now, the main part that is left is actually creating the gym environment, with a reward system, and setting that up with a PPO algorithm. We should have a basic working system in the next two weeks if all goes well.

willzoo, 2025-03-31:
Totally forgot to write a comment last week (oops), but as Danush said we finished writing all the publishers and subscribers we need so far, mainly the Imu subscriber and the cmd_wrench (movement) publisher. Writing the gym environment and making a reward system is the next step. To make the reward system we'll need to either reward based on collision or based on the position of the sub (whether it has reached or collided with a target or not).

mohana-pamidi, 2025-04-01:
Met last week to figure out what we need to subscribe to/publish to for our action states. Added code for our observation states (for camera/imu data). To get reliable imu data for the sub, we need to subscribe to the filter IMU after the Kalaman filter, so we were able to figure where we can get that data from. Next steps this week is to finish working on clearly defining action and observation spaces.

willzoo, 2025-04-06:
Update: I met with Mohana this week and we got some work done on setting up the environment. We added action spaces for the movement of the submarine and connected it with the movement publisher, and we also added a file to get data from the IMU to the gym environment. We are currently using files to communicate between publishers subscribers and the environment, but it is a little annoying having to reconstruct each message from byte data so it might have been better to have all the pubs and subs running in a thread in the gym environment file.

danushsingla, 2025-04-06:
Was unable to meet with the team this week and contribute code due to exams. I have been looking into ways the project can be expanded through other techniques besides PPO (such as TRPO or GRPO RL algorithms) and have been reading up on research for their implementation.

mohana-pamidi, 2025-04-06:
This week, we started constructing the thread to read IMU data while calculating reward. Currently, to calculate reward, I am using the percentage of red in the image we receive from the camera as the indicator for the reward. Once we are done with the threading to read in IMU data and publish then we can start training/testing.

danushsingla, 2025-04-13:
Edited the code with comments and organized the code segments so that it looks cleaner. With the comments, I am hoping to outline a clear path that we can follow to expediate the development phase and get some working model before the semester is over.

willzoo, 2025-04-14:
Although we were not able to meet in the lab this week, I did some work over the weekend, and coded a thread based approach to transfer data from subscribers to the RL environment, as opposed to our current method of using files. Although having intermittently running subscriber threads in the sub_env will take up cpu time, it is easier to implement, and is probably faster than file io. I also familiarized myself with Danush's comments and the buoy detector script.

mohana-pamidi, 2025-04-14:
I was able to work this weekend and I'm currently almost done with software for the environment for the RL model. The next step would be to write that script that trains it. I will try to start working on this script this week. I also looked at the comments created in the repo for the outline of our goals.

mohana-pamidi, 2025-04-20:
This week, I was able to meet with the rest of the team to finish integrating any software we worked on separately and start testing the environment file. Initially we were running the IMU and camera data collection on separate threads, however, we realized at the last meeting that this may not be needed. So we will have to work on that next week. For our reset() function, we tested it and were able to get it to relaunch Gazebo so the sub can spawn at its original location again. We need to continue testing rest of the functions like the reward by placing a red buoy in the Gazebo world.

willzoo, 2025-04-20:
Met with Mohana and Danush this week to integrate environment code we wrote separately. During the meeting we could not get the IMU and camera data collection to work on separate threads, however, I did some work on my own time and fixed the issue by adding a MultiThreaded Executor. Additionally I fixed the issue where the Gazebo failed to open after running ros2 launch in a terminal spawned by a python script, by resetting some environment variables before programmatically launching the terminal. Now that reset() is mostly working, we should finally be able to start implementing the reward system this week.

danushsingla, 2025-04-21:
Met with the team to continue working on the main environment code. We had some trouble with the multithreading that is still being figured out. But we did setup most of the environment to the point where the next step should be testing to ensure the environment works and then working on building the RL model.

danushsingla, 2025-10-20:
Spent a lot of time getting the remote ssh to work with the lab's computer, but it's so laggy it's hard to run gazebo on it. I'm going to do some more testing and figuring out if there is a better way to get things to run. Otherwise, I plan on going to the lab tomorrow so I can run it there.

danushsingla, 2025-10-26:
Things have worked out and I am running the simulation, but there are some problems with getting the buoy to show. I did add stabe baselines3 to the requirements.txt file and made some other minor changes for reproducability. The issues with the buoy is still in a work in progress but I hope to get that done by next week so that training can show.

danushsingla, 2025-11-09:
I've fixed a bunch of the pathing issues and it was an easy push, but there were pre-commit checks added that prevented me form properly pushing. Some of them were codespell issues that were incorrect, so I had to spent some time coordinating with Carlos so that I could push again.