uf-mil/mil2 #96: Test suite for subjugator_localization

alexoj46, 2025-03-28:
Write a suite of tests to check that the odometry output from subjugator_localization matches the real/expected approximate location of the sub. For now, focus on use with gazebo simulation, but should also be able to use with real sub pretty interchangeably.

NeedToLookAtMyPictures, 2025-03-28:
As of 3/28/2025, I closed the following issue https://github.com/uf-mil/mil2/issues/80 ,  addressed package issues with my VM setup, and begun work on the above issue.

alexoj46, 2025-04-02:
For starting localization:
- ros2 launch subjugator_localization subjugator_localization.launch.py
- ros2 topic echo /subjugator_localization/odometry/filtered

For movement:
- thruster manager should launch automatically with gazebo
- ros2 run subjugator_keyboard_control subjugator_keyboard_control (or similar? use tab complete to find)
- OR can directly publish to /cmd_wrench to move

NeedToLookAtMyPictures, 2025-04-07:
I began the week by addressing issues with the simulation (I came to the lab for a fix). I wanted to start localization work during the weekend but wasn't able to. I was hampered by finals prep (frontloading work) and weekend events (DSI symposium, etc.).

closed 2026-08-27 by Carlosdc25