uf-mil/mil2 #98: Change the Simulated Thrusters to be one message

Keith-Khadar, 2025-03-30:
The thruster board expects the messages for the thrusters to be in one message not multiple.

It should be in one message type ThrusterEfforts.msg.

CarterKreis, 2025-04-07:
- Mapped out necessary ROS and Gazebo nodes that need to be created with Keith
 - Added plugin to CMake and Gazebo World File
 - Created ThrusterBridge.cc/hh with a basic outline of the function and variables needed
 - Lots of gazebo errors so far which I hope to fix in the next day or two
 
My goal is to finish this plugin by the end of this week

alexoj46, 2025-04-07:
note on this: currently have a basic node called forward_to_sim inside the thruster_manager package that is doing the conversion from 1 to 8 messages on the ROS side only; when you are done this can be deleted :)

CarterKreis, 2025-04-14:
- Fixed the fun Gazebo errors
- Learned more about how Gazebo and ROS2 topics and nodes work
- Pull Requested this plugin which looked to work well from my testing
- Now we can publish to the ROS2 topic 'thruster/efforts' and Gazebo will properly send the values to each thruster :)

Also, hi Daniel!

CarterKreis, 2025-04-18:
The scaling factor from ROS2 topic /thruster_efforts to Gazebo is pretty small and funky so Alex and Keith added some things that broke it from my side so I couldn't work on it lol, but I did some MILpedia work and explained it on the issue

CarterKreis, 2025-04-27:
I did not work much in the lab this week due to final exams and projects, but I looked at the thruster plugin with Keith. He explained to me ROS parameters and git stash which was hiding his useful code. 

When running the plugin with the proper thruster parameters, the thrusters moved far more than before but on the wrong axis. I went back and tried to copy exactly how alex had implemented the thruster-bridge, but that did not fix the issue. Hi daniel