uf-mil/mil2 #99: Add tests for `{software,bot}@mil.ufl.edu`

cbrxyz, 2025-03-30:
We should add CI tests to ensure that our account credentials (or at least the ones that are stored in CI) are still operational and still allow us access to the accounts.

TheRickyZhang, 2025-03-31:
I can take this. Could I get more detail about the what the specific accounts are and access to them mean? I searched for @mil.ufl.edu in the codebase and couldn't find anything except for the legacy docs building.

cbrxyz, 2025-03-31:
You can just submit a PR that adds a GitHub action check that checks whether the inbox of an arbitrary SMTP account can be seen.

You can add it to [`.github/workflows/healthcheck.yaml`](https://github.com/uf-mil/mil2/blob/main/.github/workflows/healthcheck.yaml). It should take in a GitHub secret for the account server, email address, username, password, etc.

The check succeeds if the number of messages in the inbox can be fetched, and fails if not (ie, an auth/networking error has occurred).

TheRickyZhang, 2025-04-07:
I added the account-access command in healthcheck.yaml. Since I can't access the secrets I cannot test this out, but the account information is expected to be in the JSON form:
{
  "accounts": [
    { "name": "acc1", "server": "smtp1.example.com" },
    { "name": "acc2", "server": "smtp2.example.com" }
  ]
}
(this organizes for possibility of multiple accounts, but if we only have static few we can use single variables).
Please let me know when you update the secrets, and I can verify from there.