uf-mil/mil2 #140: Write Documentation for Creating Gazebo Plugins

Keith-Khadar, 2025-04-13:
Write documentation for creating gazebo plugins and put it on our wiki so that future members can quickly learn about gazebo and how to make plugins.

oixel, 2025-04-14:
I have created an [entry](https://milwiki.cbrxyz.com/wiki/Gazebo_Plugins) in the MILpedia about creating Gazebo plugins.

Additionally, I have taken [notes](https://milwiki.cbrxyz.com/wiki/PID_Controller) on PID controller lectures I have been watching and have started transporting them to MILpedia.

CarterKreis, 2025-04-18:
I added a MILpeida page explaining some of the functions within the _lovely_ Gazebo: [Gazebo Functions](https://milwiki.cbrxyz.com/wiki/Gazebo_Functions)

I also proofread and added a little bit to Oixel's page.