uf-mil/mil2 #146: Update Gazebo Models to Match New Competition Spec

Keith-Khadar, 2025-04-16:
Change the textures in the models and you might have to change the mesh of some.
Here is the rule book: https://robonation.gitbook.io/robosub-resources/section-3-autonomy-challenge/3.2-task-descriptions

oixel, 2025-04-18:
Started working on this in separate [branch](https://github.com/uf-mil/mil2/tree/updated-tasks).

oixel, 2025-04-18:
The models are all finished in the "updated-tasks" [branch](https://github.com/uf-mil/mil2/tree/updated-tasks). For some reason launching the simulator is broken on main, so I am unable to edit their imports/positions in the world; i will try tweaking it from a different branch later though.

oixel, 2025-04-28:
Have not been able to work on this due to finals / final projects. Additionally, my simulator is refusing to launch even on a cleaned up clone of the repository, so I have spent the free time I had trying to fix that. The models are finished, they just need to be put in the correct positions.