uf-mil/mil2 #256: Migrate Kill Board

DaniParr, 2025-09-15:
## Describe the feature
Migrate the navigator kill board to ROS2: https://github.com/uf-mil/mil/tree/master/NaviGator/hardware_drivers/navigator_kill_board 

## Describe the solution you'd like**
Ensure that the solution is in C++.

mklingamsetty, 2025-10-04:
Pushed onto Migrate_Kill_Board branch. Same message was said in the push below:

Add rough C++ translation of kill board simulator (draft - needs testing)

This commit translates the Python kill board simulation code to C++ as a first pass at migrating the Navigator's hardware simulation components. The translation includes all three main classes (NoopSerial, SimulatedSerial, and SimulatedKillBoard) with complete protocol handling, ROS service integration, and serial buffer management. While the code compiles and maintains the same functional structure as the original Python implementation, it's still a rough draft that needs thorough testing before integration. The implementation preserves all the original hex protocol communication, timeout checking, and kill state management, but requires validation against actual Navigator hardware drivers and comprehensive unit testing to ensure compatibility with the existing ROS ecosystem.

mklingamsetty, 2025-10-15:
I pushed with the updated format that you wanted it in. Let me know if you want me to start writing any documentation or additional comments in the code, or just anything additional needed.