uf-mil/mil2 #258: Migrate NaviGator Gazebo Simulation

DaniParr, 2025-09-15:
## Describe the feature
Migrate https://github.com/uf-mil/mil/tree/master/NaviGator/simulation/navigator_gazebo to ROS2 and the corresponding simulation environment.

## Describe the solution you'd like**
See how we migrated our SubjuGator simulation environment and do so similarly with the ROS1 NaviGator simulation.

ranjivsajay, 2025-09-22:
I reviewed the existing ROS1 simulator (navigator_gazebo) and the SubjuGator ROS2 migration as a reference. I Verified local ROS 2 workspace and repo access.

zrzz-hq, 2025-09-22:
This issue is a part of navigator ros2 migration project. @ranjivsajay Could you join the 'nav-ros2-migration' discord channel ?