uf-mil/mil2 #264: Simulation: Create plugin to actuate gripper

Carlosdc25, 2025-09-15:
## Task

Create a gazebo plugin that moves the gripper. 

**Note:** The gripper will first need to be made in simulation, but someone can still look into and make tests on this.

## Specifications

- The gripper should be able to fully open and close.
- We should be able to actuate the gripper by publishing to a ROS2 topic.

CarterKreis, 2025-09-20:
This [link](https://gazebosim.org/api/sim/9/jointcontrollers.html#:~:text=For%20controlling%20joints%20one%20would,version=%221.0%22%20?%3E) from Gazebo Sim might be useful/helpful for understanding way to move a Joint via a plugin

sgupta1703, 2025-10-05:
Hi, can I work on this issue?

sgupta1703, 2025-10-18:
Made this so far for the GripperController.cc. Studied the marble dropper and basically replicated but with the joints instead of dropping marble. Need to add 'u' to the keyboard controller to see if it will work. Idk if this works even =)

```
#include "GripperControl.hh"

#include <gz/plugin/Register.hh>
#include <gz/common/Console.hh>

namespace gripper_control
{

GripperControl::GripperControl()
{
}

GripperControl::~GripperControl()
{
}

void GripperControl::Configure(gz::sim::Entity const &entity, std::shared_ptr<sdf::Element const> const &sdf,
                               gz::sim::EntityComponentManager &ecm, gz::sim::EventManager & /*eventMgr*/)
{
    std::cout << "[GripperControl] Configuring..." << std::endl;

    if (!rclcpp::ok())
    {
        rclcpp::init(0, nullptr);
    }
    this->node_ = std::make_shared<rclcpp::Node>("gripper_control_plugin_node");

    // Subscribe to keyboard presses (std_msgs::msg::String)
    this->key_sub_ = this->node_->create_subscription<std_msgs::msg::String>(
        "/keyboard/keypress", 10, std::bind(&GripperControl::KeypressCallback, this, std::placeholders::_1));

    auto nameComp = ecm.Component<gz::sim::components::Name>(entity);
    if (nameComp)
    {
        this->model_name_ = nameComp->Data();
        std::cout << "[GripperControl] Got model name from Entity: " << this->model_name_ << std::endl;
    }
    else if (sdf && sdf->HasAttribute("name"))
    {
        try
        {
            this->model_name_ = sdf->Get<std::string>("name");
            std::cout << "[GripperControl] Got model name from SDF: " << this->model_name_ << std::endl;
        }
        catch (...)
        {
            gzerr << "[GripperControl] Failed to get model name from SDF attribute 'name'." << std::endl;
        }
    }
    else
    {
        gzerr << "[GripperControl] WARNING: Could not determine model name; plugin topic may be invalid." << std::endl;
    }

    if (sdf)
    {
        if (sdf->HasElement("joint_name"))
        {
            this->joint_name_ = sdf->Get<std::string>("joint_name");
        }
        if (sdf->HasElement("open_pos"))
        {
            this->open_pos_ = sdf->Get<double>("open_pos");
        }
        if (sdf->HasElement("closed_pos"))
        {
            this->closed_pos_ = sdf->Get<double>("closed_pos");
        }
    }

    // Build Gazebo joint command topic
    if (!this->model_name_.empty())
    {
        // topic pattern: /model/<model_name>/joint/<joint_name>/cmd_pos
        this->topic_name_ = "/model/" + this->model_name_ + "/joint/" + this->joint_name_ + "/cmd_pos";
        std::cout << "[GripperControl] Advertising gz topic: " << this->topic_name_ << std::endl;
        this->gz_pub_ = this->gz_node_.Advertise<gz::msgs::Double>(this->topic_name_);
    }
}

void GripperControl::KeypressCallback(std_msgs::msg::String::SharedPtr msg)
{
    if (!msg)
        return;

    if (msg->data == "u" && !this->u_pressed_)
    {
        this->u_pressed_ = true;
        std::cout << "[GripperControl] 'u' pressed (queued toggle)" << std::endl;
    }
}

void GripperControl::PostUpdate(gz::sim::UpdateInfo const &info, gz::sim::EntityComponentManager const & /*ecm*/)
{
    if (info.paused)
        return;

    rclcpp::spin_some(this->node_);

    // If a 'u' press was detected, toggle and publish the new joint command
    if (this->u_pressed_)
    {
        this->gripper_open_ = !this->gripper_open_;
        double cmd = this->gripper_open_ ? this->open_pos_ : this->closed_pos_;

        if (!this->topic_name_.empty())
        {
            gz::msgs::Double msg;
            msg.set_data(cmd);
            this->gz_pub_.Publish(msg);
            std::cout << "[GripperControl] Published cmd_pos = " << cmd << " to " << this->topic_name_
                      << " (gripper_open=" << (this->gripper_open_ ? "true" : "false") << ")" << std::endl;
        }
        else
        {
            gzerr << "[GripperControl] Cannot publish - topic name empty." << std::endl;
        }

        // Reset one-shot flag
        this->u_pressed_ = false;
    }
}

}  // namespace gripper_control

GZ_ADD_PLUGIN(gripper_control::GripperControl, gz::sim::System, gz::sim::ISystemConfigure,
              gz::sim::ISystemPostUpdate)
```

sgupta1703, 2025-11-10:
### Next Step: Testing the GripperControl Plugin in Gazebo

The next step is to test the `GripperControl` plugin in Gazebo to ensure the gripper joint responds correctly to key inputs. Specifically:

1. **Key Detection**: Press the `u` key and verify that `PostUpdate` correctly sets `u_pressed_` to `true`.
2. **State Toggle**: Ensure that the gripper state (`gripper_open_`) toggles between open and closed positions when `u` is pressed.
3. **Command Publication**: Check that the command position (`cmd`) is published to the intended topic (`gz_pub_`) and received by the joint controller.
4. **Joint Movement**: Confirm that the gripper joint in Gazebo moves to the expected positions (`open_pos_` or `closed_pos_`).

**Notes / Potential Adjustments**:

- Verify that `topic_name_` matches the joint controller subscription.
- Ensure `open_pos_` and `closed_pos_` are within the joint's limits.
- If movement is delayed or inconsistent, check the update loop timing and consider throttling the command publication.
- Add logging or debugging as needed to confirm message flow and joint response.

Once the gripper responds correctly, the plugin can be further expanded to handle multiple inputs or more complex control logic.

sgupta1703, 2025-11-17:
I’m going to test the plugin end-to-end to make sure the joint actually responds when the `u` key is pressed.

### What I’ll Verify
- **Key Detection:** Confirm that pressing `u` triggers `KeypressCallback` and sets `u_pressed_ = true`.
- **State Toggle:** Check that `gripper_open_` toggles correctly inside `PostUpdate`.
- **Command Publishing:** Make sure a `gz::msgs::Double` with the correct position is published to  
  `/model/<model_name>/joint/<joint_name>/cmd_pos`.
- **Joint Movement:** Confirm that the gripper joint in Gazebo moves to `open_pos_` or `closed_pos_` as expected.

### Quick Repro Commands

Send a keyboard message manually:
```bash
ros2 topic pub /keyboard/keypress std_msgs/msg/String "{data: 'u'}" -r 1

CarterKreis, 2026-02-13:
Worked with Sarthak today to finalize some things with the Gripper and its plugin. We fixed the laughably large gripper arms to be more like a bulldozer claw. We got the grippers to hold onto objects and release them. It wasn't the smoothest release but we could always just increase the opening width by changing a variable. Grabbing objects felt nearly impossible. The claws actuate and hold stuff. Not too sure what more we could add to this

Here's a demo video (thanks to camera man Dean)
[Demo Video](https://youtube.com/shorts/X7sjr3g0X3s?feature=share)

sgupta1703, 2026-02-23:
I reckon we should add some sort of friction to the gripper arms in the future because it would allow them to grip the bottle that you need to grip at the bottom of the tank.

closed 2026-04-22 by CarterKreis