uf-mil/mil2 #266: Simulation: Connect Gripper Arm and Base with Gazebo Joints

CarterKreis, 2025-09-20:
# Background
- Once the Gripper Arm and Base have been designed, we need to connect them together in Gazebo with a Joint.
- Additionally, we need to duplicated the Gripper Arm so that there are two arms which will eventually move together

# Specifications
- A Gazebo Joint should be used to connect one Arm to the Base and another Joint to connect the second Arm to the Base

# Testing
- Add the combined Gripper Arms and Base object into a Gazebo World and verify that they look connected and have a Gazebo Joint that is shared between them

 **Example**
[image]

William-Dunlap, 2025-09-26:
[image]
I added the gripper base to the 2024 gazebo world so that I can attach the arms when they are ready.

CarterKreis, 2025-11-02:
Here is a link to a forum which explains a method to scale a model within the <mesh> tag: [link](https://robotics.stackexchange.com/questions/83403/how-to-resize-meshes-in-urdf)